fzi-forschungszentrum-informatik/cartesian_controllers #152: Problem with colcon build

Daan0103, 2023-10-23:
I want to build the cartesian_controller in my workspace in 'src' but when i do colcon build, a CMake Error hits.
I use ROS2 humble on ubuntu (22.04).

This is the error:

CMake Error at CMakeLists.txt:11 (if):
   if given arguments:

        "STREQUAL" "humble"
  
   Unkwown arguments specified.

[image]

Someone who can help me :) ?

Daan0103, 2023-10-24:
Hi,
Thanks for the comment!
My output for ` echo $ROS_DISTRO ` is `humble`

Daan0103, 2023-10-24:
Is there another way to fix this, or can i do something else to implement this 'cartesian_controllers'?

stefanscherzinger, 2023-10-25:
@Daan0103 

That is really weird. It looks like the error from [this post](https://stackoverflow.com/questions/39707772), which is caused by an empty variable string if the `ROS_DISTRO` env variable is not set. But this does not seem to be your case..

I would delete the complete `build` folder from your workspace, open an new terminal there, then
```bash
source /opt/ros/humble/setup.bash
colcon build --packages-skip cartesian_controller_simulation cartesian_controller_tests --cmake-args -DCMAKE_BUILD_TYPE=Release
```
and see if the error is the same.

Daan0103, 2023-10-30:
@stefanscherzinger 

Hmm, unfortunately still the same error...

Daan0103, 2023-11-01:
Is there another way I can fix this error, is only a problem for 'Humble'?
I'm improving a previous project, where this package is used, with ROS2 Foxy, Ubuntu 20.04 LTS.
I want to go further with this project on a PC with Ubuntu 22.04 LTS and ROS2 Humble.

When creating a workspace, everything works excepts the cartesian controller where this error occurs...

gavanderhoorn, 2023-11-01:
@stefanscherzinger: not a real solution (as in: it doesn't explain *why* the current implementation is failing), but you could probably avoid the CMake-trickery and use something like `RCLCPP_VERSION_MAJOR` (defined in `rclcpp/version.h`). That's already a `#define`, so no need to do something like that yourself.

That file is generated at build time by `ament_generate_version_header(..)` which takes its data from the package's manifest.

For Iron fi, `RCLCPP_VERSION_MAJOR` is `21` (from [here](https://github.com/ros2/rclcpp/blob/613bcc52ba7c4ebc6a795991fb2c1c2a5b9e2108/rclcpp/package.xml#L5)), while for Humble, it's set to `16` (from [here](https://github.com/ros2/rclcpp/blob/c1bf0d382eaf6dfe8429020b07611ffb6d552a97/rclcpp/package.xml#L5)).

There's even a `RCLCPP_VERSION_GTE(..)` macro which should make comparing versions a little easier.

Daan0103, 2023-11-01:
Hi,
I 'played' something with my 'old' workspace. This was build in a previous project (which I have to finish on another PC). 
My PC runs on Ubuntu 22.04 LTS and ROS2 Humble, the previous PC on 20.04 and Foxy.
I split the packages to all different test-workspaces to see in which package the problem was, here the cartesian_controller build successfully. The problem is in another package in combination with newer packages/updates like this.

So, the cartesian controller is built successfully and now I have to solve the problem by other packages ;)
Anyway, thanks for the help I appreciate it!

closed 2023-11-01 by Daan0103