fzi-forschungszentrum-informatik/cartesian_controllers #172: cartesian motion controller leads to bizzare movement

shrutichakraborty, 2024-02-13:
Hello @fzi-admin @fmauch , 

I am using the cartesian motion controller with ur10e robot. after launching the robot and switching the controller, the robot makes some abrupt small motions even though nothing is commanded. Is this expected behaviour? 

Also I want to know which frame should the target_pose be in? I am assuming it should be the same frame as the end efffector. 

Thanks!

stefanscherzinger, 2024-05-04:
@shrutichakraborty 
I'm picking up older issues, sorry for not responding earlier.

> after launching the robot and switching the controller, the robot makes some abrupt small motions even though nothing is commanded. Is this expected behaviour?

This sounds a little to what I experienced in [this issue](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/142). Is this still happening today?

> Also I want to know which frame should the target_pose be in?

The `target_frame` is with respect to the specified [robot_base_link](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/dc803377752bcbc175004a549485ef011ec4a952/cartesian_controller_simulation/config/controller_manager.yaml#L43). The controllers will complain if that frame is not specified correctly in the message's header.

shrutichakraborty, 2024-05-06:
> @shrutichakraborty I'm picking up older issues, sorry for not responding earlier.
> 
> > after launching the robot and switching the controller, the robot makes some abrupt small motions even though nothing is commanded. Is this expected behaviour?
> 
> This sounds a little like what I experienced in [this issue](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/142). Is this still happening today?
> 
> > Also I want to know which frame should the target_pose be in?
> 
> The `target_frame` is with respect to the specified [robot_base_link](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/dc803377752bcbc175004a549485ef011ec4a952/cartesian_controller_simulation/config/controller_manager.yaml#L43). The controllers will complain if that frame is not specified correctly in the message's header.

Hi @stefanscherzinger, thanks for your response, after tuning the error_scale and iterations parameters I was able to get a setup where there is no longer bizzare movements on switching controller; However, when commanding positions, I still have the issue where I see the robot vibrate (or wobble) around the end effector once it reaches (or gets close to) the commanded pose. I don't know if the issue here can be resolved by tuning the parameters but I have so far not found any combination that avoids this behaviour..  Any suggestions?