fzi-forschungszentrum-informatik/cartesian_controllers #173: nothing published to /cartesian_force_controller/current_pose

shrutichakraborty, 2024-02-13:
Hi! First of all the controllers work great :) 
I am using the cartesian controllers with ros2 and ur10e robot.  I have a control loop that will continue to publish to the /cartesian_force_controller/target_wrench until a certain threshold of force. However during this time while the controller runs nothing seems to be published to /cartesian_force_controller/current_pose topic. Any ideas would be appreciated!

stefanscherzinger, 2024-05-04:
@shrutichakraborty 
Sorry for not responding earlier. I'm assuming you made it work concerning your other issues on this repository. If still relevant, could you quickly post your insights here so that others can benefit from what you found? Many thanks :)

closed 2024-05-04 by stefanscherzinger

lxk-221, 2024-06-17:
hi @shrutichakraborty , i have the same problem with you, but i'am using ros1, Ubuntu 18.04 and ros melodic.
i can use rosrun `tf tf_echo frame1 frame2` to get the pose of `ee_link`, and by publishing target pose to `target_frame_topic` of the cartesian_compliance_controller, i can control the pose of ee. However, nothing published to `\current_pose` topic.
I'd appreciate if you can give me some suggestions!