fzi-forschungszentrum-informatik/cartesian_controllers #175: getting feedback/result when trajectory is competed by robot

shrutichakraborty, 2024-03-07:
Hello all! The controllers work well. However, I would like to know if there is  a way to get some feedback/result to know when the the robot has reached the desired target_frame that we publish to the cartesian motion controller? I am looking  for somthing similar to the approach in scaled_joint_trajectory_controller where we get a result if the trajectory was done and if there were any issues in executing it? 

If not can you suggest a workaround? Thanks a lot!

matthias-mayr, 2024-03-07:
The same repo that I linked in #176 can do that for you in ROS 1.  

It spawns an action server that takes a Cartesian goal. In the [configuration file](https://github.com/matthias-mayr/cartesian_trajectory_generator/blob/master/config/trajectory_generator_params.yaml) one can specify maximum thresholds for it to mark the goal as achieved.  
These need to be chosen carefully as compliant controllers often have deviations and it might never arrive.

shrutichakraborty, 2024-03-19:
> The same repo that I linked in #176 can do that for you in ROS 1.
>
> It spawns an action server that takes a Cartesian goal. In the [configuration [...]

Thanks @matthias-mayr , however, I am using ros2 humble, do you know of any solutions that could be used here?

matthias-mayr, 2024-03-19:
I have not looked in how to do this with ROS 2 yet.  
If the helper controllers in this repo are ported to ROS 2, one solution can be to use MoveIt for linear trajectories and/or the Pilz motion planner in MoveIt. In that case, MoveIt would also take care of the completion, however with a compliant controller it needs to be parameterized accordingly since there are deviations that can make it abort.

Or you port the trajectory generator to ROS 2. It's fairly modular and it should not be necessary to touch the core. Unfortunately I will not have time to do it myself.

stefanscherzinger, 2024-04-13:
@shrutichakraborty 

If still relevant for your, you could monitor the controllers' feedback by enabling the [publish_state_feedback](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/resources/doc/Solver_details.md#solver-parameters) parameter.
However, since all controller inputs are plain topics, there's no direct mechanism for informing the caller about reaching the goals, as there is with _trajectory action_ based controllers. In addition, when working in compliance mode, it's not straight-forward to decide whether goals are reached because there's a tolerated offset due to interaction with the environment.

A workaround would be implementing a small monitor node that decides whether targets are _sufficiently well_ reached.

closed 2024-04-13 by stefanscherzinger