fzi-forschungszentrum-informatik/cartesian_controllers #176: Controlling the robot velocity in cartesian_motion_controller

shrutichakraborty, 2024-03-07:
Hello! I would like to know if there is any way of managing the speed with shich the the robot moves when we publish to cartesian_mostion_controller/target_frame? For instance I have to make very precice movements with lowered speed, how could I do that?

matthias-mayr, 2024-03-07:
It would move slower for lower stiffness values, but what is typically actually needed is some controlled way of moving the reference point.

For example one can use motion planning frameworks like MoveIt and some of the helper controllers in this repo.  
For our experiments I wrote a simple one that just publishes linear Cartesian trajectories with fixed acceleration & deceleration as well as some maximum velocity: https://github.com/matthias-mayr/cartesian_trajectory_generator It is currently only available for ROS 1 though.

shrutichakraborty, 2024-03-26:
> It would move slower for lower stiffness values, but what is typically actually needed is some controlled way of moving the reference point.
>
> For example one can use motion planning frameworks like [...]

Hi Matthias! Thanks for your response. Does your framework work on top of the Cartesian controllers? For instance, is it going to take a target_pose and basically interpolate between the current and the target_pose and publish the intermediate poses to the cartesian_motion_controller?

matthias-mayr, 2024-03-27:
Yes, that's exactly what it does. We used it for example in [this paper](https://arxiv.org/abs/2308.14206) with the `cartesian_controllers` package on a `UR5e` and a [Cartesian impedance controller](https://github.com/matthias-mayr/Cartesian-Impedance-Controller) running on an `iiwa`. Eventually the trajectory generator only published a reference pose, so whatever controller takes that as an input can be a target.

stefanscherzinger, 2024-05-04:
@shrutichakraborty

If you need this feature for ROS2 and are willing to code , here's a [promising fork](https://github.com/personalrobotics/cartesian_controllers/commit/e54ef815f2ea62822ef5d54b4a7414d93ae3a3ba) by @egordon that implements a `cartesian_twist_controller`. I'm happy to take PRs :).