fzi-forschungszentrum-informatik/cartesian_controllers #178: tuning force controller

shrutichakraborty, 2024-03-18:
Hi! I'm using the cartesian force controller. I've noticed that when sending a traget wrench in purely z (aiming for the robot to move vertically down while keepng perpendicular to the surface), it starts to drift in x and y as well. Can anyone give a hit on which parameters of the controllers could I tune to avoid this ? In my application I am checking the sensor wrench values and commanding the robot to continue to move downwards in z until it comes in contact with the surface (experiences a certain force threshold)? I am using a UR10e robot.

captain-yoshi, 2024-03-22:
Can you post your controller's configuration ?

Be sure to read the [force controller README](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/cartesian_force_controller/README.md) :

> Note that the controller does not strictly move only in the commanded direction. Although its behavior is linearized in operational space, there might be small drifts in other axes. This is a feature of the forward dynamics-based solver. In fact, there is no error reduction on axes orthogonal to the published target wrench. The benefit is that the controller finds approximate solutions near singular configurations.

This is a feature of the force controller. @stefanscherzinger Would it be possible to setup a maximum permitted position deviation per axe ? If so, could you point out where we could make this modification ?

Lowering the P Gains for the X/Y axis might help lowering the motion in those directions. There might be other causes creating the drift, e.g. drift in your force torque sensor.

shrutic12, 2024-04-08:
> Can you post your controller's configuration ?
>
> Be sure to read the [force controller [...]

`cartesian_force_controller:
  ros__parameters:
    end_effector_link: ft_frame
    robot_base_link: base_link
    ft_sensor_ref_link: ft_frame
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    
    solver:
        error_scale: 0.5
        publish_state_feedback: True

    pd_gains:
        trans_x: {p: 0.05}
        trans_y: {p: 0.05}
        trans_z: {p: 0.05}
        rot_x: {p: 0.01}
        rot_y: {p: 0.01}
        rot_z: {p: 0.01}`

Hi @captain-yoshi , I am using the default values for the force controller. 

Also I am having issues with the cartesian compliance controller: changing the stiffness values of the axes, does not seem to make any difference on the robot (ur10e). It tries to go to commanded position and does not stabilize and stop at obstacles. Any ideas sould be greatly appreciated! Here is the config for the controller (same as default), I have tested stiffness as low as 90 in all directions. 

`stiffness:  # w.r.t. compliance_ref_link coordinates
        trans_x: 500.0
        trans_y: 500.0
        trans_z: 500.0
        rot_x: 20.0
        rot_y: 20.0
        rot_z: 20.0

    solver:
        error_scale: 0.5
        iterations: 1
        publish_state_feedback: True

    # For all controllers, these gains are w.r.t. the robot_base_link coordinates.
    pd_gains:
        trans_x: {p: 0.05, d: 0.005}
        trans_y: {p: 0.05, d: 0.005}
        trans_z: {p: 0.05, d: 0.005}
        rot_x: {p: 1.5}
        rot_y: {p: 1.5}
        rot_z: {p: 1.5}`

stefanscherzinger, 2024-05-04:
@captain-yoshi  

> This is a feature of the force controller. @stefanscherzinger Would it be possible to setup a maximum permitted position deviation per axe ? If so, could you point out where we could make this modification ?

This would require some form of error rejection. The `cartesian_compliance_controller` could be used for this. However, I like the idea of _locking_ individual Cartesian axes.. That would require a new solver and a new thinking though. Not sure where to put it at the moment..

@shrutic12
>  It tries to go to commanded position and does not stabilize and stop at obstacles. 

It the robot doesn't stop, then probably the signals from the force-torque sensor are not connected to the controller. I suggest moving the robot by hand and _feel_ if the robot is responsive. If it isn't, check whether the relevant topics are published and subscribed correctly.
```bash
ros2 topic list | grep sensor
```

closed 2024-10-18 by stefanscherzinger