fzi-forschungszentrum-informatik/cartesian_controllers #179: [Iron] - Robot description is empty if ros2control is configured to load it from topic

firesurfer, 2024-03-19:
Hi @stefanscherzinger,

ros2control deprecated loading the robot_description from the parameter file and wants us to load it from the corresponding topic. I just adapted to this change and realized I can not load the cartesian controllers anymore. 

For example for the `cartesian_motion_controller` 

```
[ros2_control_node-12] [INFO] [1710834336.674188673] [controller_manager]: Loading controller 'cartesian_motion_controller_top'
[ros2_control_node-12] [INFO] [1710834336.701249189] [controller_manager]: Configuring controller 'cartesian_motion_controller_top'
[ros2_control_node-12] [ERROR] [1710834336.703519230] [cartesian_motion_controller_top]: robot_description is empty
[ros2_control_node-12] [WARN] [1710834336.703674259] [cartesian_motion_controller_top]: Error occurred while doing error handling.
[ros2_control_node-12] [ERROR] [1710834336.703766080] [controller_manager]: After configuring, controller 'cartesian_motion_controller_top' is in state 'unconfigured' , expected inactive.

```

Not sure if this is a bug in ros2control (that is should properly pass this parameter to the controller nodes) or if a controller is supposed to obtain the robot_description from the corresponding topic itself. 
I will also open an issue in the ros2control repository asking how this should be handled.

firesurfer, 2024-03-20:
@stefanscherzinger 
A follow up for you -> In the current rolling branch of ros2control they introduced the URDF as member of the controller base class. 
This change will not be backported to iron/humble.

tonynajjar, 2024-08-19:
So what's the latest state? How do we actually load a controller from this repo if the robot_description is read from the topic?

firesurfer, 2024-08-20:
You simply don't at the moment. I think @stefanscherzinger doesn't find any time to work on these controllers at the moment.

stefanscherzinger, 2024-10-20:
@firesurfer @tonynajjar 

This should now be solved with a mixed approach (see #198 ):
- **Jazzy**: Load from topic as recommended. See e.g. [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/f53f798029ea5e05993e8a5d1736c713379af0e2/cartesian_controller_simulation/launch/simulation.launch.py#L97). The cartesian controllers will fetch the `robot_description` directly from the controllers' base class.
- **Iron, Humble**: Additionally pass the `robot_description` to the `controller_manager`. See e.g. [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/f53f798029ea5e05993e8a5d1736c713379af0e2/cartesian_controller_simulation/launch/simulation.launch.py#L93). The cartesian controllers will get the `robot_description` from the `controller_manager` as parameter (depricated but working)

stefanscherzinger, 2024-10-20:
Please re-open if still unclear.

closed 2024-10-20 by stefanscherzinger