fzi-forschungszentrum-informatik/cartesian_controllers #186: segmentation fault

nihal1964, 2024-04-11:
I have tried to develop a controller to move along x,y & z axis. for which i tried for which i tried to use the cartesian controller base package. but for some reason i even after i build my code i run into a segmentation fault. i have treid to change the code in various ways. could anyone help me out with this. i will add my code below.

#include "/home/nihal/hc_ur10e/src/arrow_keys_control/include/arrow_keys_control/arrow_keys_control.hpp"
#include <iostream>
#include <chrono>
#include <atomic>
#include <signal.h>
#include <unistd.h>
#include <termios.h>
#include <sys/ioctl.h>
#include <cmath>
#include <mutex>
#include "cartesian_controller_base/Utility.h"
#include "controller_interface/controller_interface.hpp"

#define KEYCODE_Q 0x71
#define KEYCODE_A 0x61
#define KEYCODE_W 0x77
#define KEYCODE_S 0x73
#define KEYCODE_E 0x65
#define KEYCODE_D 0x64
#define KEYCODE_X 0x78

namespace arrow_keys_control
{
    ArrowKeysControl::ArrowKeysControl(rclcpp::NodeOptions options)
        : Node("arrow_keys_control_node", std::move(options)),
          Base::CartesianControllerBase(),
          key_thread_running_(true),
          m_cartesian_input(ctrl::Vector6D::Zero())
    {
    }

    rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn ArrowKeysControl::on_init()
    {
        try {
            const auto ret = Base::on_init();
            if (ret != rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn::SUCCESS) {
                RCLCPP_ERROR(get_logger(), "Failed to initialize base class");
                return ret;
            }
            auto_declare<std::string>("ft_sensor_ref_link", "");
            auto_declare("timer", "true");
            auto_declare("step_size", STEP_SIZE);
        
        } catch (const std::exception& e) {
            RCLCPP_ERROR(get_logger(), "Exception occurred in on_init(): %s", e.what());
            return rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn::ERROR;
        }
        
        return rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn::SUCCESS;
    }

    rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn ArrowKeysControl::on_configure(
        const rclcpp_lifecycle::State &previous_state)
    {
        try {
            const auto ret = Base::on_configure(previous_state);
            if (ret != rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn::SUCCESS) {
                RCLCPP_ERROR(get_logger(), "Failed to configure base class");
                return ret;
            }

            // Declare parameters
            m_ft_sensor_ref_link = get_node()->get_parameter("ft_sensor_ref_link").as_string();
            if (!Base::robotChainContains(m_ft_sensor_ref_link))
            {
                RCLCPP_ERROR(get_logger(), "%s is not part of the kinematic chain from %s to %s", 
                              m_ft_sensor_ref_link.c_str(), Base::m_robot_base_link.c_str(), Base::m_end_effector_link.c_str());
                return rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn::ERROR;
[... 192 lines cut ...]
} // namespace arrow_keys_control

int main(int argc, char** argv) {
        rclcpp::init(argc, argv);
        rclcpp::NodeOptions node_options;
        node_options.use_intra_process_comms(true);
        node_options.automatically_declare_parameters_from_overrides(true);
        auto node = std::make_shared<rclcpp::Node>("arrow_keys_control", node_options);

        // Create the ArrowKeysControl node
        auto arrow_keys_control = std::make_shared<arrow_keys_control::ArrowKeysControl>(node->get_node_options());
        arrow_keys_control->configure();

        // Trigger the activate state transition
        arrow_keys_control->on_activate(rclcpp_lifecycle::State());

        // Register SIGINT handler
        signal(SIGINT, arrow_keys_control::exitSigHandler);

        // Spin the node
        arrow_keys_control->spin();

        rclcpp::shutdown();
        return 0;
}

stefanscherzinger, 2024-04-13:
HI @nihal1964 

Thanks for asking this. It's a little outside the scope of this ROS1/2 controller suite, but let's see..

Could you run your _executable_ inside gdb (GNU debugger), wait for the _segfault_ and post the output of the `backtrace` command here?
Here's a [short introduction](https://github.com/stefanscherzinger/gdb_best_practices/blob/main/slides.md#common-workflow-with-ros1ros2) I once created for working with _gdb_ in the ROS1/2 context.

nihal1964, 2024-04-15:
So I have tried valgrind, but the issue is it shows tons of issues. I probably think it is trying to follow the rclcpp functions on my code and the rclcpp functions from Cartesian_controller_base and there is a possibility that they are accessing the same memory which could cause an issue if it is already full or deleted

nihal1964, 2024-04-15:
The only problem is I have no idea how to get through this

stefanscherzinger, 2024-05-04:
@nihal1964
If this is still relevant for you, could you post some error messages? I'm particularly interested in _gdb's_  output of `backtrace` after the segfault. I think that's easier than using valgrind here.

Other than that, if you are interested in controlling a robot with keyboard commands and the `cartesian_controllers` package, I recommend you implement something on top of the `cartesian_motion_controller` and use its `geometry_msgs/PoseStamped` interface. Here's [an example](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/dc803377752bcbc175004a549485ef011ec4a952/cartesian_controller_utilities/scripts/pose.py#L114) where I do that with incremental input in a Python node. You could do something similar and interpret keystrokes as velocities.

nihal1964, 2024-05-04:
so, as far as i remember it basically refers to segmentation fault due to buffer overflow. i tried a little about changing the memory allocations but then it shows a lot of build errors.
As for cartesian_motion_controller, i have used that by creating a node for keyboard input with each key updating different pose & published it to the topic target_frame. which worked for a simulation perfectly, i have to test on the real robot though. 
but at this point i am just curious about the segmentation fault

stefanscherzinger, 2024-05-06:
> As for cartesian_motion_controller, i have used that by creating a node for keyboard input with each key updating different pose & published it to the topic target_frame. which worked for a simulation perfectly, i have to test on the real robot though.

Alright, sounds great.

> but at this point i am just curious about the segmentation fault

I could try to have a look. Could you either fork the _cartesian controllers_, add your changes, or put your code into a new Github repo to share?

nihal1964, 2024-05-06:
Sure, I will do that.
Also I have one question. I recently tested the node I created for Cartesian motion controller. But all the moment is with the base frame as reference. For example, for the first move if I say move 0.5 along+ve x axis, the end effector moves to a position which is 0.5 from the base frame. After the first move it moves according to the current position but that is not what we need right, if I say move 0.5 along +ve x axis I would normally mean the end effector to move from its current position 0.5 along+ve x axis.

nihal1964, 2024-05-10:
[cartesian_controller(2).zip](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/files/15276365/cartesian_controller.2.zip)
sorry for the delay

stefanscherzinger, 2024-10-18:
Unfortunately, I didn't find the time to have a look at it. I'm closing this here. The recommended way is building upon the `cartesian_controller_base` with a proper `ros2_control`-based controller.

closed 2024-10-18 by stefanscherzinger