fzi-forschungszentrum-informatik/cartesian_controllers #191: difference in /current_pose publishe dby the three controllers

shrutic12, 2024-04-15:
Hi! I am using the cartesian controllers with ros2 humble and ur10e robot. I was going to remap the `cartesian_motion_controller/current_pose`, `cartesian_compliance_controller/current_pose` and `cartesian_force_controller/current_pose` topics to the same topic `/current_pose` to make subscription easier. But I noticed on doing an echo on the three topics that the current pose is different for the three topics for the same robot position without moving it at all. 

[image]

The
The real values from the teach pendant are : 
x: -375.27 (mm)
y: 168.96
z: 397.74 
RX : 2.224 (rad)
RY : 2.219 (rad)
RZ : 0.0 (rad)

Any ideas why this could be happening?

stefanscherzinger, 2024-05-04:
Thanks @shrutic12 for pointing this out.

I think there might be three things happening that cause your observation:
1. Each UR robot should use a specific _calibration_ file. I'm not sure how that works today, but there were differences between what the _teach pendant_ says and what the _ROS2 side_ publishes if this calibration is not used. Unfortunately, I can't test at the moment if the driver reports its current pose. At least, [this issue](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/571) is still open. It would be interesting to check what _the ROS2 UR driver_ says where it is.
2. The `cartesian_controller`s IK solver [uses the joint positions](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/dc803377752bcbc175004a549485ef011ec4a952/cartesian_controller_base/src/IKSolver.cpp#L134) that it gets from the robot. However, it also runs an internal control loop to consider external inputs, such as measured force-torque values from the sensor. Depending on sensor noise, there might be small numeric differences in the sub millimeter range. The `cartesian_motion_controller` should be closest to what the ROS2 driver says, because it does not consider forces.
3. The `current_pose` depends on each controller's configuration of the [end_effector_link](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/dc803377752bcbc175004a549485ef011ec4a952/cartesian_controller_simulation/config/controller_manager.yaml#L37). I assume that the controller from your second image has a different link specified. That's why its orientation is different than the other two.

stefanscherzinger, 2024-06-04:
Please reopen if still unclear

closed 2024-06-04 by stefanscherzinger