fzi-forschungszentrum-informatik/cartesian_controllers #205: How Can I Use Cartesian Compliance Controller on Real UR5e Robot?

Yeesha-R, 2024-07-27:
Hi, 

I am new to ROS and python programming and I have some questions regarding these controllers:
1. How can I launch this package to work on a real UR5e robot?
2. Can I use the cartesian compliance controller when the robot is in freedrive mode?
3. I want my robot to be less stiff when in the freedrive mode. What parameters of the controller (stiffness or pd_gains) do I need to tweak to achieve this?

I look forward to your help.
Thank you so much.

stefanscherzinger, 2024-08-19:
Hi @Yeesha-R 

Thanks for your interest in using this controllers suite!

1. Here's a small [getting started package](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots) for the Universal Robots.
2. No, this won't be possible. The UR5e's _freedrive mode_ will deactivate all ROS controllers that would write commands to the hardware to avoid conflicting commands.
3. As a consequence of point _2._ I'm afraid this is out of scope of these controllers and needs to be adjusted somewhere in the UR's parameter range. I'm not sure, however, if the e-Series has the sensors and internal control algorithms to shape apparent inertia as do some of the high-end collaborative robots.

Yeesha-R, 2024-08-19:
Thank you so much for your response @stefanscherzinger . I have been able to implement the controller.

closed 2024-08-19 by Yeesha-R