fzi-forschungszentrum-informatik/cartesian_controllers #206: /cartesian_motion_controller/current_pose not publishing

Yazan-Soliman, 2024-08-01:
Hello, 
I am working on a project on a ur10e robot where i have to move it in Cartesian space. 
while it is easy to move the robot to a desired target_frame over /cartesian_motion_controller/target_frame , subscribing to /cartesian_motion_controller/current_pose is not returning any data. I needed this so that the publisher waits until the robot has reached its target.

Thanks in advance!
**Software versions**
 - **ROS version**: Humble

leledeyuan00, 2024-08-18:
Have you checked the parameter "publish_state_feedback" on configuration yaml file is setted to "on"?

Yazan-Soliman, 2024-08-19:
@leledeyuan00  actually it is.

```[INFO] [1724062107.393394715] [parameter_lister]:   publish_state_feedback: True```

stefanscherzinger, 2024-08-19:
@Yazan-Soliman 

That's interesting. I'm not sure about the `[parameter_lister]`, but could you again check the parameter via the controller directly
```bash
ros2 param get /<your-controller-name> solver.publish_state_feedback
```
Note that the topic only gets published once the controller is _active_. 

Additionally, what is the controller's update rate that you are running your setup with? The UR10e should mention that somewhere on the command line output during startup.

stefanscherzinger, 2024-10-20:
Please re-open if still unclear.

closed 2024-10-20 by stefanscherzinger