fzi-forschungszentrum-informatik/cartesian_controllers #207: A setup problem with ROS1-Noetic

clowz1, 2024-08-21:
When I run the rosdep command “rosdep install --from-paths ./ --ignore-src -y”， it shows the error that:

ERROR: the following packages/stacks could not have their rosdep keys resolvedto system dependences：XXXXXXX Cannot locate rosdep defintion for [ament_cmake]

I searched this error on the website, it means that I didn't install ament_cmake, but I can't install it with command, please tell me how to solve it, my ros version is noetic, thanks.

gavanderhoorn, 2024-08-21:
`ament_cmake` is a package used in ROS 2.

Can you make sure you checked out the `ros1` branch of this repository?

clowz1, 2024-08-21:
Yes，I am sure that I did the command from the ROS1 branch and run these two command “git clone ***@***.***:fzi-forschungszentrum-informatik/cartesian_controllers.git
rosdep install --from-paths ./ --ignore-src -y”

。。
***@***.***

gavanderhoorn, 2024-08-21:
What's the output of `git rev-parse HEAD`? Run it in your clone of this repository.

clowz1, 2024-08-21:
dc803377752bcbc175004a54985ef011ec4a952

。。
***@***.***

gavanderhoorn, 2024-08-21:
That's `HEAD` on the `ros2` branch: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/commit/dc803377752bcbc175004a549485ef011ec4a952.

You need to `git checkout ros1`. Then try again.

clowz1, 2024-08-21:
Thanks for the reply. I understand what this problem means.

 
That's HEAD on the ros2 branch: dc80337.

stefanscherzinger, 2024-09-26:
Thanks @gavanderhoorn for taking care. @clowz1 Please re-open if still unclear.

closed 2024-09-26 by stefanscherzinger