fzi-forschungszentrum-informatik/cartesian_controllers #208: Random Jerk/trembling during execution with cartesian_motion_controller

shrutichakraborty, 2024-09-02:
**Problem description**
Hi authors, 

I have been using the all the controllers, tested both on ros1 noetic with UR10E and on ROS2 humble version with a different UR10E robot, for a while now for different use cases and have experienced some issues that I wanted to raise that might help make future improvements to the controllers. 

1. Random Jerk/trembling during execution with cartesian_motion_controller 

 - In my use case I require long continuous use of the cartesian_motion_control (about an hour) during which the robot uses a external attached camera to perform some processing and then move accordingly. I have noticed that the further the distance between the robot's current and target destination, the faster the robot moves towards the target destination and the start is not smooth, i.e the robot make an abrupt movement to move towards the destination. For this I have tried to tweak the P gains for the controller in XY direction and the error_scale but I still notice the same problem. When the target point to move is closer to the current robot position the motion is acceptable and the robot moves without a sudden jerk at start. 
 
I am wondering if this has been noticed by others before and if there is any advice to avoid this behaviour? I am also wondering if there is a possibility to allow users to control the velocity of the motion which might be like a scale factor to avoid big jerky motions? 

 -  Another issue I have faced is, often during the image processing part of my pipeline, the cartesian_motion_controller is still active and the external control on the robot is active too butthe robot is still. In this case with increased duration of running, I see the robot make sudden random jerks (even when no commands are being sent ).. I am not sure why this occurs and a fix would be greatly helpful as currently this tends to affect my image processing pipeline as the camera images come out blurry due to the robot having randomly moved. 
 
 - Finally, with the cartesian_motion_controller, if I stop the robot launch file, without first stopping external control , I notice that the robot starts moving randomly. I have been taking care to properly switch the cartesian_controller off and cutting of extrenal control to the UR before stopping the launch file on the PC but still this is a strange issue and  a fix would be really helpful. 
 
3. Precision error with cartesian_compliance_controller 
- I tested the commanding a set of known position to the robot in compliant mode to check its precision. On comparing the position reaced by the robot vs the position commanded to the the robot I have noticed an error in precision of about 2-5 cm in the 3 axes. Is this something to be expected? 

- Additionally, I also tried using the controller for teleoperating in compliance mode. I am sending a wrench (computed from positions and velocity of my teleop device) and the current robot position at each timestep for the teleop. However, in certain cases the robot tries to come back to its original starting position. This was also the case with the example teleop scripts with the space_nav mouse, as soon as we let go the robot returns to the center to its original start position. Whereas, what I would like is for the robot to remain where it was teleoperated to.. Anny suggestions how I could ensure this behaviour? 

I had also tried to teleop by only sending positions and that did not work as mentioned in my previous posts #194 

Here is my config for the controllers : 

```
cartesian_motion_controller:
  ros__parameters:

    # See the cartesian_compliance_controller
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint

    # See the cartesian_compliance_controller
    command_interfaces:
      - position
        #- velocity

    solver:
        error_scale: 0.9
        iterations: 12
        publish_state_feedback: True

    pd_gains:
        trans_x: {p: 1.0}
        trans_y: {p: 1.0}
        trans_z: {p: 1.6}
        rot_x: {p: 0.5}
        rot_y: {p: 0.5}
        rot_z: {p: 0.5}
```
for the cartesian_compliance_controller I am using the base config given in the example with UR robot except the `error_scale` is at 0.25 as my robot vibrates at anything higher. 

Thanks a lot!

stefanscherzinger, 2024-10-21:
Thank you @shrutichakraborty for taking the time and giving thorough feedback!

I appreciate your insights and will try to work that into future improvements. Some of which are also on the documentation side.
Additionally, I'll try to add a little on selected points and make helpful comments.

_1. Random Jerk/trembling during execution with cartesian_motion_controller_
   > I have noticed that the further the distance between the robot's current and target destination, the faster the robot moves towards the target destination and the start is not smooth

   Yes, that is correct. All controllers do interpolate towards the targets by themselves, but depending on the parameters, this can result anywhere from fast jumps to a smooth motion. The first steps towards a fresh target are always the _most jerky_ ones, because the controllers reduce the error iteratively. That being said, you can obtain a fast and smooth behavior at the same time. It's all determined via the target frame itself. Instead of publishing individual, distant target frames, publish your target frame as a function of time with a velocity profile of your choice and at a rate of e.g. 100 Hz. Then parameterize the `cartesian_motion_controller` to follow the target accurately.

   >  I see the robot make sudden random jerks (even when no commands are being sent ).. I am not sure why this occurs and a fix would be greatly helpful

   I completely agree. This should not be the case. However, can you prove that it's not some artifact coming from the image processing pipeline causing this jerky motion?

   > if I stop the robot launch file, without first stopping external control , I notice that the robot starts moving randomly.

   This should also not be the case. Does that happen sometimes or always? Does it happen also with other non-`cartesian_controllers` controllers?

_3. Precision error with cartesian_compliance_controller_

   >  I have noticed an error in precision of about 4-7 mm in the 3 axes. Is this something to be expected?

   It depends. The `cartesian_compliance_controller` models a spring-like behavior in which it relates measured forces/torques to the distance from a target. If the controller measures something in some direction, then it will be off from the target in this direction. How much it is off can be specified with the stiffness parameters.

   > Whereas, what I would like is for the robot to remain where it was teleoperated to.. Anny suggestions how I could ensure this behaviour?

   I think it's more intuitive to teleoperate the robot with the `cartesian_force_controller` which I also explained in #194. The `cartesian_compliance_controller` is more for commanding the robot via a script, like what you do with the `cartesian_motion_controller` but where you expect some contact with the environment.

shrutichakraborty, 2024-10-22:
Hi @stefanscherzinger ,
Thanks a lot for taking the time to get back to this issue and for all your comments! 

> I completely agree. This should not be the case. However, can you prove that it's not some artifact coming from the image processing pipeline causing this jerky motion?

My pipeline works like this:

1. Move the robot to a location using cartesian_motion_controller
2. Grab an image from camera and process, return a set of positions for the robot to visit
3. For each of the target_positions, move the robot to them using the cartesian_motion_controller and perform some operations using the force controller. 
4. At the end, grab image and repeat until no objects of interest detected from the camera. 

I notice the random jerks during the step 2, these are also more noticable when the robot has been running for longer hours. During this step I have verified that no target_frames are  being published to the robot during this time. 
I guess a hack to solve it will be to deactivate the controller once it has reached to the target locations but that would be starting and stopping the controller several times. If there is another olution it would be great. 

> 

> > if I stop the robot launch file, without first stopping external control , I notice that the robot starts moving randomly.
> 
> This should also not be the case. Does that happen sometimes or always? Does it happen also with other non-`cartesian_controllers` controllers?

This happens all the time if I stop the launch file of the robot, either without deactivating the controller or stopping external control. No this only happens with the cartesian_controllers especially the force and motion controller. 

Thanks!
Looking forward to hearing from you!