fzi-forschungszentrum-informatik/cartesian_controllers #209: joint position check for controller using velocity interface

304x, 2024-09-04:
## Problem statement and motivation
I am using the cartesian_motion_controller with the velocity command interface. I am working on a Franka Emika Panda robot and have noticed that the robot throws the following error for some target poses: **"joint_motion_generator_position_limits_violation"**
I have looked at the source code and if I understand correctly when using the velocity interface the joint limits read from the robot model are not considered before sending the velocity commands which leads to the mentioned error. 

## Goal of this feature
No target pose leads to an invalid joint position. 

## Approach and next steps
Maybe a way to archive this is to clip the velocity commands before sending them to ensure that they do not lead to an invalid joint position just as it is done with the position commands.

stefanscherzinger, 2024-10-18:
@304x 

Yes, you are right. This would be a good feature. I think I saw that on a fork and will try to get a PR for it.

InigoMoreno, 2025-03-11:
Any update on this? I am getting similar issues, could you share the fork you mentioned so I can have a look?