fzi-forschungszentrum-informatik/cartesian_controllers #215: Controlling UR5e

garg-akash, 2024-10-26:
Hello,

Firstly thank you for providing this interesting repository. 

I have been playing around with the simulations. I have some basic questions, sorry if they are too obvious.
1. Inside the computeError() in the cartesian force controller ([here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/cartesian_force_controller/src/cartesian_force_controller.cpp#L144)) the error is computed like this [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/cartesian_force_controller/src/cartesian_force_controller.cpp#L159). Shouldn't the error be difference between the target and the ft sensor value?

2. Where does the gravity compensation come from? The joint accelerations are computed using the net force [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/cartesian_controller_base/src/ForwardDynamicsSolver.cpp#L88).

3. I have a UR5e in my lab with External Control Urcap installed. How can I use the force controller from this repository to control the real robot?

Thank you,
Akgr

stefanscherzinger, 2024-10-26:
Hi @garg-akash 

Thanks for your feedback!

> 1. Shouldn't the error be difference between the target and the ft sensor value?

This controller implementation follows a little different philosophy. It uses forward dynamics of an external net force to the end-effector and simulates how the robot moves according to that net force. The simulated motion is then sent to the joints of the real system. The net force is what the sensor measures from possible collisions/interaction with the environment and what users additionally specify as target wrench. If you are interested, I explained that a little more in detail [in this pdf](https://publikationen.bibliothek.kit.edu/1000139834) in chapter 4. 

> 2. Where does the gravity compensation come from? 

There is not gravity compensation currently. We assume that you zero (bias, tare, etc.) the force torque sensor on the robot before going into contact, so that everything you measure is only external contacts from interaction with the environment.

> 3. How can I use the force controller from this repository to control the real robot?

There's a getting started section with useful links [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2?tab=readme-ov-file#getting-started). Hope that helps.

stefanscherzinger, 2024-10-26:
Please re-open if something is still unclear.

closed 2024-10-26 by stefanscherzinger