fzi-forschungszentrum-informatik/cartesian_controllers #219: How to use cartesian_compliance_controller with MoveIt2 and real UR10e robot?

Adrian2702, 2025-01-14:
**Problem description**

I use the **ur_robot_driver** to control a real UR10e robot. The robot gets new poses from a C++ script. Everything works fine with the **scaled_joint_trajectory_controller** from **ur_controllers**. But for the application (sanding over an object) it is necessary to consider the contact forces with the object and not to use pure position control as before. So I tried the **cartesian_compliance_controller**. As far as I have implemented it, it is possible to activate the controller and send pose data like: 

`ros2 topic pub --once /target_frame geometry_msgs/msg/PoseStamped '{'header': {'stamp': 'now', 'frame_id': 'base_link'}, 'pose': {'position':{x: 0.5, y: 0.5, z: 0.5}, 'orientation':{x: 0, y: 0, z: 0, w: 1}}}'`

 to the robot. So far I use MoveIt2 to plan the trajectory. However, when I now send my poses via C++ script with **cartesian_compliance_controller** enabled, the **move_group.moveit.moveit.plugins.simple_controller_manager** throws an error because it only recognizes 2 controllers (_scaled_joint_trajectory_controller and joint_trajectory_controller_) as configured in the **moveit_controllers.yaml** file in the **ur_moveit_config** package. My question is, how can I continue to use MoveIt2 as my planner but in combination with the cartesian_compliance_controller? There was some kind of workaround for ROS1, but is there still a good solution for ROS2?

**Software versions**
 - **Commit**: f8aad77697802d9b9ca08d044b87ce58167e4bf9
 - **OS**: Linux 6.8.0-45-generic
 - **ROS version**: ROS2 Jazzy

edtriccorp, 2025-01-21:
Hi @Adrian2702,

Were you able to get the Cartesian controllers to work/launch with Jazzy? I ran into issues when trying to launch them with Jazzy. I have them working in Humble. I also recently started investigating how to get MoveIt2 integrated with the Cartesian Controllers. 

The MoveIt2 [move_group_interface_tutorial.cpp](https://github.com/moveit/moveit2_tutorials/blob/e0974b62a3b8ec06c224ffd66d7ce4a9a147b2a9/doc/examples/move_group_interface/src/move_group_interface_tutorial.cpp) has basic examples of Cartesian Path Planning but I am unclear on how to connect MoveIt2 to the Cartesian Controller /targetframe topics. We are also interested in constraining the joint configuration of the eoat ... for example parallel to a surface ....etc... I am unclear if this can be achieved by Cartesian Path planning or if it needs to be hybrid. Similar to the example section "Enforce Planning in Joint Space".

leledeyuan00, 2025-03-04:
> Hi @Adrian2702,
> 
> Were you able to get the Cartesian controllers to work/launch with Jazzy? I ran into issues when trying to launch them with Jazzy. I have them working in Humble. I also recently started investigating how to get MoveIt2 integrated with the Cartesian Controllers. 
> 
> The MoveIt2 [move_group_interface_tutorial.cpp](https://github.com/moveit/moveit2_tutorials/blob/e0974b62a3b8ec06c224ffd66d7ce4a9a147b2a9/doc/examples/move_group_interface/src/move_group_interface_tutorial.cpp) has basic examples of Cartesian Path Planning but I am unclear on how to connect MoveIt2 to the Cartesian Controller /targetframe topics. We are also interested in constraining the joint configuration of the eoat ... for example parallel to a surface ....etc... I am unclear if this can be achieved by Cartesian Path planning or if it needs to be hybrid. Similar to the example section "Enforce Planning in Joint Space". 
> 
> I am gathering so far that it appears that using MoveIt2 would result in IK solving which kind of defeats the strength of these controllers but I am hoping my understanding is wrong? 
> 
> 

Hi @edtriccorp, You're right, Moveit realized the IK by its IK plugins. And then generating a message for joint trajectory controller. I think the way is to write a new moveit plugin that using the cartesian kinematics. I tried it before, but the performance I think is not good. Just for your reference.